xaelsouth/rtl-wmbus #1: SSE4 alternative?

papadeltasierra, 2017-09-18:
The code as is does not compile on a PC without SSE4 support.  My slightly older PC only supports SSE3 and I've got the code running but I was surprised that there is only #ifdef SSE4, #endif and not #else arms.  Are you able to give a high level of what the SSE4 code is doing in case it's worth my writing some equivalent SSE3 code?

xaelsouth, 2017-09-20:
In the Makefile remove "-msse4.2"  for both debug and release targets.

xaelsouth, 2017-09-20:
Sorry, i have not read all the details - you have got the code already working...

#else branch is not of any interest. All that SSE4 does - iterate over vectors of samples each of length of 8.  If absence of SSE4 the subtraction would be performed for __every__ sample, not for vector.

papadeltasierra, 2017-09-21:
OK, thanks for taking the time to respond.

closed 2018-01-23 by xaelsouth