xaelsouth/rtl-wmbus #6: CRC-issues

sm3ulc, 2019-12-12:
Seem to have some strange issue with CRC errors. The "10030325" is a Kamstrup flowIQ 2200 and I'm pretty sure the IDs are the same. Have tried with three different dongles including a rtlsdr-v3 and some combinations of antennas. Have received different levels of RSSI but seem unable to get a single packet with OK CRC.  Any ideas?

C1;0;1;2019-12-12 17:40:03.000;8;7;10030332;0x3644372c320303103a163108f01920a85417a08312020a019d8f8af8cfad73221b05c33dfc6618a93c5a6277d1de1d5d071f8242f959c1
C1;0;1;2019-12-12 17:40:28.000;117;117;10030325;0x3644372c250303103a16ff00f01920c49674902ae43a6f88c416e528d90a0bb8bcea9de6b949f3a6b4d54183c8c99c136e707694a659c1
C1;0;1;2019-12-12 17:42:52.000;8;8;10030330;0x3644372c300303103a164320f01920afd1ead56723e577f2faf31e9fd6b860a3f5e6e78fe9847d997fd7defec2483df3e0c4e43dc159c1
C1;0;1;2019-12-12 17:43:46.000;113;116;10030325;0x3644372c250303103a160138f01920b27d02632ce5a00a92d9686d9c13ad578b579cc79c637a89724fe757bd59de898bfa469b4c0159c1
C1;0;1;2019-12-12 17:45:49.000;29;32;10030266;0x3644372c660203103a164c14f41920c4c93344c164e0f85a5fe9ba080e1f7e08c6ca62052e6437057af458c2f0ec2ab2756a20fe2b59c1

weetmuts, 2019-12-13:
Check out the fork https://github.com/afflux/rtl-wmbus.git it can receive C1 telegrams.

sm3ulc, 2019-12-13:
Got perfect reception right away! Thanks!

makuser, 2020-03-26:
I have the same problem with a Lorenz water meter, which is a rebranded [Engelmann Waterstar M](https://www.engelmann.de/de/waterstar-m-2/).
```
C1;0;1;2020-03-26 07:12:27.000;120;128;19062913;0x9644fa121329061900077aef000020046d0c289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c4031300000000000413000000008004130000000080028980000000620289800000000003098000008062030980000000420389800000004003898000000042040980000000
C1;0;1;2020-03-26 07:12:58.000;134;129;19062913;0x9644fa121329061900077af0000020046d0c289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c4021300000000420189800000006201898000000004028980000000c4020980000000420144c0800000310144c00000002100c2600000e01880c2600000001080e2600000001140e26000000010810260000000
C1;0;1;2020-03-26 07:13:29.000;130;140;19062913;0x9644fa121329061900077af1000020046d0d289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c40213000000004201898000000062018980000000400144c0000000220104c0000000020144c0800000310144c00000002003098000008062030980000000420389800000004003898000000042040980000000
C1;0;1;2020-03-26 07:13:59.000;137;146;19062913;0x9644fa121329061900077af2000020046d0d289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c4031300000000800413000000008004130000000000051300000000c405130000000000061300000000c4061300000000840713000000008107130000000004081300000000
C1;0;1;2020-03-26 07:14:30.000;121;128;19062913;0x9644fa121329061900077af3000020046d0e289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c4031300000000080413000000018804130000000108051300000000c405130000000008061300000000c4061300000000840713000000008907130000000084081300000000
C1;0;1;2020-03-26 07:15:01.000;120;145;19062913;0x9644fa121329061900077af4000020046d0e289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c4031300000000800413000000008004130000000000051300000000c405130000000000061300000000c4061300000000840713000000008107130000000004081300000000
C1;0;1;2020-03-26 07:15:32.000;119;148;19062913;0x9644fa121329061900077af5000020046d0f289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c4021300000000420189800000006201898000000004028980000000c402098000000084028980000000620289800000000403098000008062030980000000420389800000004403898000000004081300000001
C1;0;1;2020-03-26 07:16:03.000;131;143;19062913;0x9644fa121329061900077af6000020046d0f289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c40213000000004201898000000062018980000000420144c0000000220104c00000004202898000000062028980000000420184c00000c0310184c00000002101c4c00000003140e26000000010408130000000
C1;0;1;2020-03-26 07:16:33.000;133;121;19062913;0x9644fa121329061900077af7000020046d10289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c40313000000000809260000000100082600000000100a2601000001880a260000000010184c0000020310184c00000002101c4c000000022738980000000410204c00000000
C1;0;1;2020-03-26 07:17:04.000;118;121;19062913;0x9644fa121329061900077af8000020046d10289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000008062600000001880626000000010009260000000300082600000000000a2601000001880a260000000010184c0000020310184c00000002101c4c00000003071c4c0000000210204c00000000
C1;0;1;2020-03-26 07:17:35.000;126;144;19062913;0x9644fa121329061900077af9000020046d11289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c403130000000084028980000000c402098000000084028980000000620289800000000403098000008062030980000000420389800000006403898000000042040980000000
C1;0;1;2020-03-26 07:18:06.000;119;134;19062913;0x9644fa121329061900077afa000020046d11289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c4031300000000000926000000018004130000000008051300000000c405130000000008061300000000c406130000000084071300000000c807130000000104081300000000
C1;0;1;2020-03-26 07:18:37.000;132;127;19062913;0x9644fa121329061900077afb000020046d12289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c403130000000084028980000000620104c0000000420144c0800000310144c0000000020184c00000c0310184c00000002101c4c00000002281c4c0000000210204c0000000
C1;0;1;2020-03-26 07:19:07.000;127;133;19062913;0x9644fa121329061900077afc000020046d12289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c4031300000000800413000000004002098000000000028980000000620289800000000003098000008062030980000000420389800000004403898000000042040980000000
C1;0;1;2020-03-26 07:19:38.000;130;138;19062913;0x9644fa121329061900077afd000020046d13289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c4031300000000800413000000008004130000000000051300000000c40513000000008403098000008062030980000000420389800000004003898000000004081300000000
[... cut ...]
C1;0;1;2020-03-26 07:45:40.000;124;127;19062913;0x9644fa121329061900077a30000020046d2d289a2304138400000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000084031300000000c4031300000000080413000000018804130000000108051300000000c40513000000008403098000008062030980000000420389800000004403898000000004081300000000
```

I already tried the fork by @afflux, the output above is from the fork.

What would you suggest I do to see where the CRC problem originates?

weetmuts, 2020-03-26:
Can you test xaelsouths latest master? It supports C1.

makuser, 2020-03-26:
> Can you test xaelsouths latest master? It supports C1.

Yes, I have originally started with that one and after reading what you wrote in https://github.com/weetmuts/wmbusmeters/issues/99, I started to look into rtl_wmbus some more, saw this issue right here and then also tried the version by afflux. I get the same output no matter what I do.

weetmuts, 2020-03-27:
@makuser Strage, do you receive other meters of the same model?

The radio decoding code is currently outside of my competence.
Perhaps you can buy an im871a or amb8465 dongle?
They seem to be high quality devices.

makuser, 2020-03-27:
> @makuser Strage, do you receive other meters of the same model?
> 
I only have one meter for testing of this model, but already requested a second one.

> The radio decoding code is currently outside of my competence.

Okay, too bad. You also don't know if there are any debug options in rtl-wmbus I could enable to see what the expected and sent CRC values are?

> Perhaps you can buy an im871a or amb8465 dongle?
> They seem to be high quality devices.

Okay, sure. I will try to source one of these. The amb8465 seems to be the better choice, am I right on that? And should it be a AMB8465 or AMB8465-M?

weetmuts, 2020-03-28:
I think it is AMB8465-M

makuser, 2020-08-02:
sooo. The AMB8465-M finally arrived on friday and I was able to run wmbusmeters with the AMB8465.

How shall I proceed best, so that we can test and fix this?

xaelsouth, 2021-02-05:
Mauser, could you test with the beweist version?

xaelsouth, 2021-02-25:
I have to consider this issue closed because of no response from users and because some fixes on C1 were made, which should improve the C1 receiver quality.

closed 2021-02-25 by xaelsouth