xaelsouth/rtl-wmbus #10: rtl-wmbus coredumps on a telegram

pswiatki, 2020-09-24:
Every time rtl-wmbus is started it crashes after a number of days. This is the backtrace:

#0  0x00007ff3df10abf4 in __memmove_avx_unaligned_erms () from /usr/lib/libc.so.6
#1  0x0000556abe0fabd6 in cook_pkt (data=<optimized out>, datalen=1) at t1_c1_packet_decoder.h:522
#2  t1_c1_packet_decoder (bit=<optimized out>, rssi=<optimized out>) at t1_c1_packet_decoder.h:606
#3  t1_c1_packet_decoder (rssi=<optimized out>, bit=<optimized out>) at t1_c1_packet_decoder.h:577
#4  main (argc=<optimized out>, argv=<optimized out>) at rtl_wmbus.c:504
(gdb) 

My guess is some "offending" datagram causes libc's `memmove() `function in `cook_pkt() `to fail.
Should it be re-created with **debug** version of the binary?

xaelsouth, 2020-10-15:
Hmm, really strange. Could you provide me a last datagram received before went crashed?

merbanan, 2021-01-26:
@pswiatki compile with -O0 -g3 then run it under valgrind. That should find the offending code. CSA might find some things also.

xaelsouth, 2021-01-26:
@pswiatki You could try to compile with "make debug", it would use switches as proposed by merbanan.

An/or give a try to the new rtl_wmbus - I have appended an 16 byte alignment attribute to the buffer on which memmove() working, but I suspect that this  not a complete solution.

pswiatki, 2021-05-13:
I no longer see the issue. Since I update the entire system every now and then (it is Arch Linux) chances are `rtl-wmbus` got out of sync with some system provided libs and after I rebuilt `rtl-wmbus` the issue never surfaced again.
I guess it can be closed now.

closed 2021-05-13 by pswiatki