xaelsouth/rtl-wmbus #13: Building on Raspberry Pi B rev 2 ARMv6

Marooned-MB, 2020-11-27:
Hi
I'm having problems compiling this package:
```
pi@raspberrypi:~ $ git clone https://github.com/xaelsouth/rtl-wmbus.git
Cloning into 'rtl-wmbus'...
remote: Enumerating objects: 58, done.
remote: Total 58 (delta 0), reused 0 (delta 0), pack-reused 58
Unpacking objects: 100% (58/58), done.
pi@raspberrypi:~ $ cd rtl-wmbus/
pi@raspberrypi:~/rtl-wmbus $ make pi1
gcc -DNDEBUG -O3 -march=armv6 -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -Iinclude -std=gnu99 -Wall -W -Waggregate-return -Wbad-function-cast -Wcast-align -Wcast-qual -Wchar-subscripts -Wcomment -Wfloat-equal -Winline -Wmain -Wmissing-noreturn -Wmissing-prototypes -Wparentheses -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-prototypes -Wswitch -Wunreachable-code -Wunused -Wuninitialized -o "build/rtl_wmbus" rtl_wmbus.c -lm
In file included from rtl_wmbus.c:37:
atan2.h: In function ‘atan2_approximation2’:
atan2.h:48:11: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
     if (x == 0.0f)
           ^~
atan2.h:51:15: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
         if (y == 0.0f) return 0.0f;
               ^~
In file included from rtl_wmbus.c:39:
t1_c1_packet_decoder.h: At top level:
t1_c1_packet_decoder.h:252:5: warning: no previous prototype for ‘get_mode_a_tlg_length’ [-Wmissing-prototypes]
 int get_mode_a_tlg_length(uint8_t lfield)
     ^~~~~~~~~~~~~~~~~~~~~
t1_c1_packet_decoder.h:257:5: warning: no previous prototype for ‘get_mode_b_tlg_length’ [-Wmissing-prototypes]
 int get_mode_b_tlg_length(uint8_t lfield)
     ^~~~~~~~~~~~~~~~~~~~~
rtl_wmbus.c:47:14: warning: ‘lp_1600kHz_56kHz’ defined but not used [-Wunused-function]
 static float lp_1600kHz_56kHz(int sample, size_t i_or_q)
              ^~~~~~~~~~~~~~~~
/usr/bin/ld: cannot open output file build/rtl_wmbus: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [Makefile:26: pi1] Error 1
```

This is a clean folder with proper rights:
```
pi@raspberrypi:~/rtl-wmbus $ ls -l
total 3340
-rwxr-xr-x 1 pi pi     476 Nov 27 19:35 androidbuild.bat
-rw-r--r-- 1 pi pi    1561 Nov 27 19:35 atan2.h
-rw-r--r-- 1 pi pi 1048576 Nov 27 19:35 demod.bin
drwxr-xr-x 2 pi pi    4096 Nov 27 19:35 filter
-rw-r--r-- 1 pi pi    2931 Nov 27 19:35 fir.h
-rw-r--r-- 1 pi pi    2377 Nov 27 19:35 iir.h
drwxr-xr-x 3 pi pi    4096 Nov 27 19:35 include
-rw-r--r-- 1 pi pi    1009 Nov 27 19:35 Makefile
-rw-r--r-- 1 pi pi    2260 Nov 27 19:35 moving_average_filter.h
-rw-r--r-- 1 pi pi     966 Nov 27 19:35 net_support.h
drwxr-xr-x 2 pi pi    4096 Nov 27 19:35 pics
-rw-r--r-- 1 pi pi    3295 Nov 27 19:35 ppf.h
-rw-r--r-- 1 pi pi    4218 Nov 27 19:35 README.md
-rw-r--r-- 1 pi pi  174065 Nov 27 19:35 readme.pdf
-rw-r--r-- 1 pi pi   18071 Nov 27 19:35 rtl_wmbus.c
-rw-r--r-- 1 pi pi 2097152 Nov 27 19:35 samples2.bin
-rw-r--r-- 1 pi pi   23336 Nov 27 19:35 t1_c1_packet_decoder.h
```

Any hint on that?
Thanks

xaelsouth, 2020-11-28:
Hi, I don't see anything errorneous in the log you have attached.  Just an idea, try to do
   mkdir build
in the folder.

Marooned-MB, 2020-11-28:
`"The best place to hide something is in plain sight."` heh
The solution was so simple that I'm ashamed I didn't think about it and had to ask about this. It worked obviously.
Thanks

closed 2020-11-28 by Marooned-MB