xaelsouth/rtl-wmbus #14: S1 packages with rtl-wmbus

raritos, 2021-01-28:
Hi there,
Can i get S1 packets with rtl-wmbus?
if yes, how?

I appreciate your help.
p.s.: i sent an email btw.

xaelsouth, 2021-01-28:
No, it's not implemented. I don't have ever seen a device with S1 mode implemented. Can you record a bitstream with devices inside?

raritos, 2021-01-28:
looks like S1 mode is the same as C/T mode, but the problem is C/T mode is 100k where as S1 is 32k, so i guess your iir filter needs to be tuned for 32k sampling.
i tried to get 868.30Mhz S1 devices thru your code, but looks like iir filter is broke with 32k sampling, since i think you tuned iir for 100k sampling , as you stated on your code, right?
if i can come up with iir filter for 32k sampling , looks like it will work, but i just guess :-)
What can i do to change your iir filter to accomodate 32k sampling.
BTW, i will get S1 bitstream

xaelsouth, 2021-01-29:
I suspect there is more of effort to invest into it. 32k sampling is just one thing; you have (self-synchronized) Manchester coding so you don't need the IIR filter for clock recovering. If you have recorded a signal from your device, i can look into it.

raritos, 2021-01-29:
Thanx xael,
---sync-patern is no problem for you :-) you already get C/T frames with  
 const uint32_t ACCESS_CODE = 0b0101010101010000111101u;
for T1 frame     n × (01) 0001110110 10010110 n ≥ 279   , you can catch it.
-- Manchester is no problem for you, i am sure.

I tried to get packages but even if i can catch frame start, trailing bits are not correct, thats why i stopped modifying your code.
you state,if i am not mistaken, that your iir filter is tuned for 100k sampling, where S1 frame is 32k, so i end up with wrong bits.

my devices have 16day rule from the start of a month, so for a couple of days, i will not be able to get any signal. after 2. of next month , i will get many samples for you to look into.
Thanx again.

xaelsouth, 2021-01-30:
You can easily recalculate bandpass filter coefficients I have used for clock recovering with  Octave, see: https://github.com/xaelsouth/rtl-wmbus/blob/master/filter/bp_iir_cheb1_800kHz_90kHz_98Hz_102kHz_110kHz.m

Ws1 = 90e3/nyqistrate;     % replace with (32768 - 10000)/nyqistrate
Wp1 = 98e3/nyqistrate;    % replace with (32768 - 2000)/nyqistrate
Wp2 = 102e3/nyqistrate;  % replace with (32768 + 2000)/nyqistrate
Ws2 = 110e3/nyqistrate;   % replace with (32768 + 10000)/nyqistrate
Rp = 1;                               % leave it so
Rs = 40;                             % leave it so

Place new coefficients in the the function bp_iir_cheb1_800kHz_90kHz_98kHz_102kHz_110kHz(float sample).

Or better you create a new function named bp_iir_cheb1_800kHz_22kHz_30kHz_34kHz_42kHz(float sample) und use this instead of old one. Provide me with the patch if it works. But i would merge this after I have seen the bitstream and done some tests with that. Good luck! ;)

alalons, 2021-02-01:
I have a test emitter from qundis capable of sending s1 data every 2 minutes. It's used to place amr collection devices and test signal strength.
If you give me the modified code I can try it anytime.

xaelsouth, 2021-02-01:
Hello alalons,

could you record the signal for a period of about 2 or more minutes with

rtl_sdr s1_samples.bin -f 868.3M -s 1600000

and share s1_samples.bin to me?

alalons, 2021-02-02:
Alvaro Alonso ha compartido un archivo de OneDrive con usted. Para verlo, haga clic en el vínculo siguiente.
<https://1drv.ms/u/s!Am-8ChbmpLqisAO5QfiMdovxbJYK>
[https://r1.res.office365.com/owa/prem/images/dc-generic_20.png]<https://1drv.ms/u/s!Am-8ChbmpLqisAO5QfiMdovxbJYK>
s1_samples.tar.gz<https://1drv.ms/u/s!Am-8ChbmpLqisAO5QfiMdovxbJYK>

alalons, 2021-02-02:
those are the telegrams being transmitted

alvaro@alvaro-Compaq-Presario-CQ61-Notebook-PC:~$ wmbusmeters --logtelegrams auto:s1
Started auto im871a[00101516] on /dev/ttyUSB0 listening on s1
Started auto rtlwmbus[00000001] listening on s1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 04998541
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x01
                device: im871a[00101516]
                  rssi: -37 dBm
                driver: unknown!
telegram=||2C4465324185990401087A008000000B6E560400426C5F0C4B6E452301326C0000046D1A0E6F0202FDAC7E0123|+6
Received telegram from: 04998541
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x01
                device: im871a[00101516]
                  rssi: -37 dBm
                driver: unknown!
telegram=||2C4465324185990401087A008000000B6E560400426C5F0C4B6E452301326C0000046D1A0E6F0202FDAC7E0123|+27
Received telegram from: 04998541
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x01
                device: im871a[00101516]
                  rssi: -37 dBm
                driver: unknown!
telegram=||2C4465324185990401087A008000000B6E560400426C5F0C4B6E452301326C0000046D1A0E6F0202FDAC7E0123|+49
Received telegram from: 04998541
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x01
                device: im871a[00101516]
                  rssi: -37 dBm
                driver: unknown!
telegram=||2C4465324185990401087A008000000B6E560400426C5F0C4B6E452301326C0000046D1A0E6F0202FDAC7E0123|+71

xaelsouth, 2021-02-02:
checkout and test

alalons, 2021-02-02:
tested ok with real meter! great work

data captured with new version
S1;1;1;2021-02-02 18:21:26.000;0;20;74117991;0x30446532917911742a087a60000000046d2d13a222025ac30001fd0c06326cffff0dff5f0c0008233000c10613080bfef9

Data captured with im871a dongle
Received telegram from: 74117991
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x2a
                device: im871a[00101517]
                  rssi: -97 dBm
                driver: unknown!
telegram=||30446532917911742A087A60000000046D2D13A222025AC30001FD0C06326CFFFF0DFF5F0C0008233000C10613080BFEF9|+353

xaelsouth, 2021-02-02:
Thanks for response! Then I will close the issue.

closed 2021-02-02 by xaelsouth