xaelsouth/rtl-wmbus #15: capturing s1 and c1,t1 simultaneously?

alalons, 2021-02-04:
I wonder if it would be possible to capture and decode all those modes simultaneously.
You could rise sampling rate to acommodate both frequencies.
Then it would be neccesary to convert to baseband in software with each center frequency creating two signals to be processed.
Or something like that.

xaelsouth, 2021-02-04:
Yes, basically this should be possible, if you would sample with a higher sample rate (2.4 MSamples instead of 1.6Msamples) and center to 868.3MHz.  Then you process S1 in the same way with only  the decimation rate pulled up to 3 (2.4MSamples) instead of 2 (1.6Msamples). But in case of T1/C1 a fequency translation must be applied, which could be done by something like FIR xlating and decimation filter (as implemented in the GnuRadio).

If I had time (in a few years or so)  iI want to implement it. Or send a patch if you want it earlier :-P.

xaelsouth, 2021-02-05:
Hello alalons,

few years were gone very quick ;), checkout and test it:

rtl_sdr -f 868.7M -s 1600000 - 2>/dev/null | build/rtl_wmbus -s

Notice here
 * 868.7M - the new rx frequency in between of 868.3M (S1) and 868.95M (T1/C1)
 * "-s" switch which informs rtl_wmbus that frequency shifting is needed

Please test and let me know if it works. I have tested for T/C1 only. Could you record a signal where both S1/C1/T1 are included?

rtl_sdr s1_T1_c1_samples.bin -f 868.7M -s 1600000

alalons, 2021-02-05:
Alvaro Alonso ha compartido un archivo de OneDrive con usted. Para verlo, haga clic en el vínculo siguiente.
<https://1drv.ms/u/s!Am-8ChbmpLqisAQYjyvhwx9yLGfD>
[https://r1.res.office365.com/owa/prem/images/dc-generic_20.png]<https://1drv.ms/u/s!Am-8ChbmpLqisAQYjyvhwx9yLGfD>
samples.tar.xz<https://1drv.ms/u/s!Am-8ChbmpLqisAQYjyvhwx9yLGfD>

Sure, time must have slowed down for the rest of us! Relativity issues 😉.
You are awesome! It works perfectly.

Ready for the next challenge?

/rtl-wmbus-master$ rtl_sdr -f 868.7M -s 1600000 - 2>/dev/null | tee s1_c1_samples.bin | build/rtl_wmbus -s
S1;1;1;2021-02-05 19:36:56.000;0;102;04998541;0x1e4665324185990401087a00800000046d1a0e6f0202fdac7e012301fd0c01
S1;1;1;2021-02-05 19:37:18.000;0;98;04998541;0x1e4665324185990401087a00800000046d1a0e6f0202fdac7e012301fd0c01
S1;1;1;2021-02-05 19:37:44.000;0;106;04998541;0x2c4465324185990401087a008000000b6e560400426c5f0c4b6e452301326c0000046d1a0e6f0202fdac7e0123
C1;1;1;2021-02-05 19:38:01.000;96;93;92966281;0x314493448162969234087a131800200b6e0000004b6e000000426c9e29cb086e000000c2086cbf21326c8c2c046d2f14a522
C1;1;1;2021-02-05 19:38:01.000;103;103;92966281;0x314493448162969234087a131800200b6e0000004b6e000000426c9e29cb086e000000c2086cbf21326c8c2c046d2f14a522
S1;1;1;2021-02-05 19:38:05.000;0;99;04998541;0x2c4465324185990401087a008000000b6e560400426c5f0c4b6e452301326c0000046d1a0e6f0202fdac7e0123

xaelsouth, 2021-02-05:
Thanks for testing and sharing the samples!

closed 2021-02-05 by xaelsouth