xaelsouth/rtl-wmbus #17: packet rssi always 0 in S1 mode

alalons, 2021-02-05:
trying to understand difference betwen packet rssi and rssi 
wmbusmeters is using packet rssi showing 0 in S1 mode.
Is it ok or needs a fix?

xaelsouth, 2021-02-05:
Not sure about 0 in S1 Mode, but rssi_filter() function should definetely be duplicated for both signal paths. Will fix this tomorrow.

xaelsouth, 2021-02-06:
1) Just fixed that the issue with rssi = 0
2) rssi_filter() splitted in two function

Please test.

alalons, 2021-02-06:
working ok now. Could you explain or link an explanation for both rssi diferences?
Thank you

alvaro@alvaro-Compaq-Presario-CQ61-Notebook-PC:~/rtl-wmbus-master$ rtl_sdr -f 868.3M -s 1600000  - 2>/dev/null | rtl_wmbus 
S1;1;1;2021-02-06 12:01:30.000;140;148;04998541;0x2c4465324185990401087a008000000b6e560400426c5f0c4b6e452301326c0000046d1a0e6f0202fdac7e0123
S1;1;1;2021-02-06 12:01:52.000;146;130;04998541;0x2c4465324185990401087a008000000b6e560400426c5f0c4b6e452301326c0000046d1a0e6f0202fdac7e0123

alvaro@alvaro-Compaq-Presario-CQ61-Notebook-PC:~/rtl-wmbus-master$ rtl_sdr -f 868.7M -s 1600000  - 2>/dev/null | rtl_wmbus -s 
S1;1;1;2021-02-06 12:02:35.000;137;141;04998541;0x2c4465324185990401087a008000000b6e560400426c5f0c4b6e452301326c0000046d1a0e6f0202fdac7e0123

xaelsouth, 2021-02-06:
It's simple. Packet rssi is an rssi value taken at the first datagram's bit (most significant bit in the L field) and will not be updated while receiving remaining bits of the current packet (or datagram). An another rssi is the current rssi which will be continuously updated with each bit received.

It's because I want to know how rssi was changing regarding to the beginning of the packet currently received.

xaelsouth, 2021-02-07:
Close

closed 2021-02-07 by xaelsouth