xaelsouth/rtl-wmbus #19: Questions about the S1,C1,T! combo mode.

weetmuts, 2021-02-11:
I saw the new support for S1 mode and have a quick question.

Does rtl-wmbus receive C1/T1 telegrams better when rtlsdr is tuned to 868.95MH (with potential fine tuning)?
Or is it equally good at receiving C1/T1 telegrams if I use the new -s option and 868.7MH ?
Likewise compared to S1 telegrams and rtlsdr tuned to 868.3M.

My intent was to write logic so that: `wmbusmeters rtlwmbus:c1,t1` 
will tune to 868.95M and  `wmbusmeters rtlwmbus:s1` will tune to 868.3M
and  `wmbusmeters rtlwmbus:c1,t1,s1` will tune to 868.7M and add -s.

But if the last c1,t1,s1 combo is just as good at receiving telegrams, then I could just 
let wmbusmeters to always default to this setting and my code will be simpler.

What do you think?

alalons, 2021-02-12:
I have been testing a new approach, see my fork. 
In this fork depending on sample rate and decimation  I have been getting different rssi values. I have enabled decimation in steps smaller than one and sample rate adaptable in multiples of 800k times decimation.
Best results with my test gear at decimation 2.7 and 2160000 Hz bandwidth with rssi 150

At decimation 1.15 and 929000Hz bandwidth rssi decreases to 80

At higher bandwidths started to decay rssi.

Xael, what do you think? Is there an optimum sample rate. Why does rssi depend (apparently) on sampling rate?

xaelsouth, 2021-02-12:
Hello Fredrik.

Well, i did not take any quantitative tests, therefore i can not _definitely_ tell in which case receiving telegrams would be better or worse. Anyway, receiving telegrams at an intermediate frequency (in our case at 868.7MHz) means that frequency translation (or shifting by multiplication) is needed. Every additional step in the signal processing chain means introducing more rounding errors, that could reduce the overall sensitivity by a few decibels. Sounds bad, but is absolutely no issue as the devices are located relatively close to receiver and there are no obstacles (like multiple concrete walls).

To my opinion you would better stick with something like `wmbusmeters rtlwmbus:c1,t1,s1:sdr@-f=868.7M,-s=1600000:wmbus@-s` which will allow users to specify command lines of rtl_sdr and rtl_wmbus as they want. You just define a best guess default value for the most common case and left further optimizations to be done by users.

xaelsouth, 2021-02-12:
Hello alalons,

To get rid of RSSI value fluctuations you probably need to normalize output from mavgi_filter() by the decimation rate. But why?..

rtl_wmbus has been designed with efficiency in mind. I don't think that decimating the sample rate as you want will be expedient.

Downsampling (and upsampling) will usually be done by an integer step: 2,3,4.... Another case like yours is basically a _resampling_ by a ratio. It's not trivial and means that first upsampling and then downsampling with additional post-filtering is needed (or first downsampling and then upsampling with pre-filtering) that will eat CPU cycles.

Decimation in rtl_wmbus is implemented as moving average filter, which is a kind of low-pass filter. This allows a very simple and efficient downsampling by an integer value, which is 2 by default. If a low-pass FIR filter would be used, it would load the CPU heavier and require filter coefficients to be recalculated for a given sample rate an every  program start.

If you faced poor performance (1xCPU system?) check following:
  - checkout newest rtl_wmbus version. I have slightly optimized in the main loop: less float additions and divisionst
  - do you really need atan() from libm or an replacement is sufficient
  - deactivate run length algorithm or comment it out. You will not need that, probably. It's actually redundant to time2 method
  - make filter using fixed-point arithmetic instead of float

And measure and measure and measure execution speed on _every_ change you make. I suggest to record the radio for one or two minutes into a file and always test rtl_wmbus fork with your modifications against. This will make your measures comparable and you will immediately see how your modifications affect execution speed and the receiver quality.

alalons, 2021-02-13:
I have no need to optimize speed right now. Just had the happy idea of computing both frequency shifts at the same time halving the number of float multiplications involved in complex computation.
as shown by the data of profiler gprof  in the pull request comments, it does halve execution time of the frequency shifting.
But as also mentioned there the main point susceptible for optimization if any possible are FIR an IIR and moving average filters wich account for most of cpu usage.

xaelsouth, 2021-02-16:
Fredrik, take the new rtl_wmbus version from today with alalon's optimizations included. Please note,  that the new frequency for simulataneous receiving of S1 and T1/C1 has moved from 868.7MHz to 868.625MHz! I will close this issue here.

closed 2021-02-16 by xaelsouth