xaelsouth/rtl-wmbus #26: Support Mode N WmBus to allow wmbusmeters project to support Honeywell TPR11 Merlin 169 WmBus

dwojciec, 2021-08-16:
rtl-wmbus supports mode T1, mode C1 and mode S1 but not mode N as mentioned here https://github.com/weetmuts/wmbusmeters/issues/321. With the Support of mode N wmbusmeters project could benefit from it.
All informations relative to Mode N (wmBus) are welcome to evaluate the complexity of packet decoding project of mode N.

merbanan, 2021-09-11:
@dwojciec can you record some signal samples? If so I would be able to add decoding support via rtl_433.

dwojciec, 2021-09-15:
> @dwojciec can you record some signal samples? If so I would be able to add decoding support via rtl_433.

Hi @merbanan thank for your support. Do you have some command you could share with me to take record of my signal ?  and where can I upload them once recorded ?

merbanan, 2021-09-15:
rtl_433 -f 169500k -s 1600000 -S unknown

That should work.

https://www.silabs.com/documents/public/application-notes/an1119-rail-wmbus.pdf

The above pdf lists the different frequencies that should be in use. The rtl_433 command should cover the whole band but you also might need to do offset tuning and then you need to tune to 169.3MHz or 169.7MHz or so.

You can also let rtl_433 try to automatically recover the bits by adding a -A command.

The sample files (*.cu8 files) you get you can attach to a github ticket. Either you open a new issue in rtl_433 or here. I will most likely add the signal samples to the rtl_433_tests signal repository. Then anyone who want to add support can use them as development base.

dwojciec, 2021-09-16:
I tried to generate files using this command :
rtl_433 -f 169728051 -s 1600000 -S unknown -A 
and 
this other command 
rtl_433 -f 169328051 -s 1600000 -S unknown -A
Result are into the zip files attached.
[file-generated.cu8.zip](https://github.com/xaelsouth/rtl-wmbus/files/7178990/file-generated.cu8.zip)

merbanan, 2021-09-17:
I looked through the files and found nothing. Can you use something like gqrx and try to locate the signal in frequency? And try and change to -S all. If possible place the decode close to the rtl-sdr hardware and then remove the antenna.

merbanan, 2021-09-17:
The device might also need to adjust the gain parameter (-g).

davemajor, 2021-10-15:
I was just about to jump in as well! I recorded some samples using the above command. It's from (I hope) my Italian water metre which outputs mode-n.

[mode n.zip](https://github.com/xaelsouth/rtl-wmbus/files/7353987/mode.n.zip)
!

merbanan, 2021-10-17:
I checked all samples and there was nothing more then noise found. I suspect that this method does not work well for signal reception. I will try some other options to see if I can get some better results.

bend94, 2022-03-31:
Hi,

My new house is almost ready. I have an ARROW WAN MVM 169. See picture 

[image]

I would like to know if it is possible de decode the signal in order to read my water consumption 
Thanks for your help
Regards

bend94, 2022-04-01:
hi
i did some research on internet.
My water counter ARROWWAN MVM 169 MHz use Frequency range: 169.400 MHz to 169.475 MHz.
I plugged a RTL2832 usb key in my raspberry pi

The RTL2832 seems to be capable of reading from 64MHz to 1.7GHz

Now, need help, looking for a procedure "how to install" software part (rtl_wmbus)
if someone can help, it could be great !
Regards

bend94, 2022-04-01:
so if i am right i have to save an I/Q-stream and decode this immediately with that command :

    rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | tee samples.bin | build/rtl_wmbus

merbanan, 2022-04-01:
rtl_433 -f 169728051 -s 1600000 -S unknown -A

Run that and post the recorded files.

bend94, 2022-04-02:
ok il will do that next week (THX)
keep you inform

ABelliqueux, 2022-04-11:
I have the same device as @bend94 (maddalena Arrow Mvm 169 ) and was able to run the command provided by @merbanan on my RTL2838 , i.e :
```
rtl_433 -f 169728051 -s 1600000 -S unknown -A
```
I got 18 *.cu8 files I'm attaching here in case it can be of use.

[maddalena_arrow_mvm_169.zip](https://github.com/xaelsouth/rtl-wmbus/files/8467430/maddalena_arrow_mvm_169.zip)

bend94, 2022-04-12:
hi sorry for the delay, my house is nearly ready. 
I plan to catch log as well
best regards

bend94, 2022-05-16:
Hi
I am back with some logs files.
I started this  commands :
`rtl_433 -f 169728051 -s 1600000 -S unknown -A`

Every 10 minutes i received something. Hope it is the water counter.
Please find the files 

[CU8.zip](https://github.com/xaelsouth/rtl-wmbus/files/8698431/CU8.zip)

Best Regards

merbanan, 2022-05-16:
I looked through both sets of files but couldn't find anything. From the manual it is specified that there is 1 reading per day in the factory settings. And there is also a Lora mode so it is unclear if any device is actually using the wmbus mode and it is hard to be sure we actually get a transmission. We might need to lower the bandwidth to properly get hold of this signal. So right now it is very difficult to advice what to do. I would need a device myself to be able to get further.

bend94, 2022-05-17:
What a shame !
do you want me to run the script for a whole day ?
do you want me to adapt the bandwith ? if yes what argument to use ?
thanks

bend94, 2022-05-17:
coming from the documentation :

Arrow MVM is available in two versions: WAN Arrow MVM 169 and Arrow WAN
The two versions differ for their transmission frequency (respectively 169 MHz and 868 MHz) and for the protocol used.
Arrow MVM 169 uses the wM-Bus transmission protocol and also conforms to AF-NOR E17Z v2.
Arrow MVM 868 uses the LoRa protocol WAN and is certified by LoRa Alliance®.
This guarantees a high level of interoperability with various reading systems on the market, including third-party systems.

My counter is a 169 version so 169 MHz and wM-Bus

merbanan, 2022-05-17:
> do you want me to run the script for a whole day ?
> do you want me to adapt the bandwith ? if yes what argument to use ?

The point is that I don't know what these would be.

Maybe try without -s 1600000 and tune to 169.4MHz directly. Maybe a tad bit lower. That should cover only 169.3MHz - 169.5MHz. Then run it for a day and see what you get. If possible remove the antenna and place it directly by the receiver.

bend94, 2022-05-17:
I saw that page on the web => https://github.com/weetmuts/wmbusmeters
do you think i can use that to decode ?
regards

bend94, 2022-05-17:
@nestorayuso  what would you run in term of command ?

merbanan, 2022-05-17:
All channel stuff doesn't matter. As long as the signal is recorded properly we can decode it. And yes just rtl_433 -f 169400000 -S unknown.

sgargel, 2022-11-28:
Hi there, I have a Gas Meter that uses W-MBUS on 169Mhz and communication protocol DLMS-COSE (not sure what does it means).

Further details here:
https://github.com/weetmuts/wmbusmeters/issues/702

Here some `cu8` samples:

[20221115.zip](https://github.com/weetmuts/wmbusmeters/files/10098517/20221115.zip)

I'm not sure how to "read" them and also I'm not sure if those signals can be in any way "decoded". 

Any help or comment will be very appreciated.

Thanks

mgambier, 2023-07-01:
Quick question for the experts on this thread. I live in the South of France and I am a Suez water company customer. In France, Suez uses the WIZE network and my counter is a Diehl/Sappel part number 3065469 which operates on the 169Mhz frequency. If my understanding is correct, the counter only sends information once or twice a day to the Suez servers. So, if this correct, I would not really gain anything to try to get the information out of my meter directly versus fetching the information from the Suez servers daily (there are a few solutions to do that around). Am I looking at this correctly? Thanks

xjols, 2024-04-09:
Hello!

I tried to decoder packets in n-Mode wireless m-bus in 169.44375MHz (channel n1d,n2d). I recorded some packets and analized without exit. This frequency is used in narrow band m-bus, and also wize protocol. Both uses GFSK at 2400bps NRZ. That is like the packet that I recorded. But analizing the protocol I can't find any of the preambles and sync bytes that the two standards refrect in papers...

Acording Specifications for m-bus mode n channel c and d:
> - Chip rate: 2.4 kbps, 2 GFSK modulation
> - Frequency deviation: ±2.4 kHz
> - Receiver Filter BW: 11.58 kHz
> - Packet Format: preamble (n = 8) x (01) + sync word “11110110 10001101” + payload + CRC
> 
Acording Specifications for wize PHY-WM2400 frames:

> Packet Format: preamble (n = 8) x (01) + sync word “11110110 01110010” + payload + CRC

I don't know how continue to identify and decode this packets. Any help ?

I attached some .cu8 captures!!!

[image]
[image]
[rtl_433_169.44_1.zip](https://github.com/xaelsouth/rtl-wmbus/files/14921219/rtl_433_169.44_1.zip)
[rtl_433_169.44375_2.zip](https://github.com/xaelsouth/rtl-wmbus/files/14921286/rtl_433_169.44375_2.zip)

DarKOrange75, 2025-05-10:
> Hi,
> 
> My new house is almost ready. I have an ARROW WAN MVM 169. See picture
> 
> [image]
> 
> I would like to know if it is possible de decode the signal in order to read my water consumption
> 
> For documentation : "Arrow WAN MVM 169 use vM-Bus protocol"
> 
> Thanks for your help Regards

I have the same meter, anyone made progress ?

bend94, 2025-05-10:
J'ai le meme compteur mais personne ne m'a donné la solution pour lire les
dataframes
et je suis interressé

bennesp, 2025-11-18:
> I've also collected some cu8 samples, my area is full of the meters from ref. [wmbusmeters/wmbusmeters#677](https://github.com/wmbusmeters/wmbusmeters/issues/677) (NexMeter RF169MHz). I captured the samples with `rtl_433 -f 169400000 -S unknown` over an approx. 24 hour timeframe. You can find the (>1GB) captures here https://drive.google.com/file/d/1J6QDhYCCJZiVGMMfPzR8hXq2aaC4TDI-/view?usp=sharing.

@optiluca Hopefully someone will find this useful [Reverse engineering Wireless M-Bus Mode N](https://bennesp.github.io/posts/007-wmbus/)

This is mostly it:

[image]