xaelsouth/rtl-wmbus #28: Kamstrup FlowIQ 2102 (Kamstrup 21) in 923 MHZ - mode C1

fc0ta, 2021-09-08:
Hello. Does rtl-wmbus can read and decode packets of a 923 Mhz Kamstrup FlowIQ 2102?  I have tried to get any packets at all and don't get any telegrams.  Could it be the sample rate?  I have been able (testing with rtl_433 with its analysis option) to get packets on 923 Mhz FSK NRZ demodulation every 16 and 96 seconds, but can't get to decode them.  Is there a way to try this in RTL-WMBUS?

Thanks in advance for reading and giving any input.

petterreinholdtsen, 2022-01-12:
I would too like to read the data from a flowmeter from Kamstrup.  Anything I can do to help implement or debug this?

This is from a test run

```
% rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | build/rtl_wmbus
T1;1;1;2022-01-12 13:08:23.000;117;145;21481969;0x1e44ae4c6919482168077a3100100504b55cfa79c3605935e04a72d015f201
```

petterreinholdtsen, 2022-01-12:
And I get this from rtl_433:

```
% rtl_433 -R 104 -R 105  -f 868900000 -s 1600000 -M pr
otocol -M level -g "" -F kv
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pere/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 2 out of 175 device decoding protocols [ 104-105 ]
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 1600000 S/s.
Tuner gain set to Auto.
Tuned to 868.900MHz.
Allocating 15 zero-copy buffers
WARNING: Undeclared field "payload_encrypted" in [104] "Wireless M-Bus, Mode C&T, 100kbps (-f 868950000 -s 1200000)"
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2022-01-12 14:45:30                    Protocol  : 104
model     : Wireless-MBus Mode      : T            Manufacturer: SEN
ID        : 21481971
Version   : 104          Device Type: 0x07         Device Type String: Water
Control   : 0x44         Data Length: 31           Data      : 1c44ae4c7119482168077a31001005fae20609d3315a9b6054a025142df9998810
Integrity : CRC
Control Info: 0x7A       Access number: 0x31       Device Type: 0x00
Configuration Word: 0x0510                         Payload Encrypted: 1
Modulation: FSK          Freq1     : 868.9 MHz     Freq2     : 869.0 MHz
RSSI      : -0.1 dB      SNR       : 27.6 dB       Noise     : -27.7 dB
```

Sadly I can not see the measurement value (should be somewhere between 1 and 2) here.

weetmuts, 2022-01-12:
@petterreinholdtsen Hi Petter, rtl_wmbus decodes the radio waves, use wmbusmeters to decode the content. :-)
Running: wmbusmeters 1c44ae4c7119482168077a31001005fae20609d3315a9b6054a025142df9998810
will print:

Received telegram from: 21481971
          manufacturer: (SEN) Sensus Metering Systems, Germany (0x4cae)
                  type: Water meter (0x07) encrypted
                   ver: 0x68
                driver: iperl
 
So your meter is already supported by wmbusmeters, however you need the key to decrypt the telegrams.

https://github.com/weetmuts/wmbusmeters/

weetmuts, 2022-01-12:
@fc0ta  You say you have collected telegrams using rtl_433? Have you had any success using
rtl_sdr -f 923.1M -s 1600000 - 2>/dev/null | rtl_wmbus
?

petterreinholdtsen, 2022-01-12:
[Fredrik Öhrström]
> So your meter is already supported by wmbusmeters, however you need
> the key to decrypt the telegrams.

Right.  Is there a way to find the key if I know what the meter is
displaying on its screen?

-- 
Happy hacking
Petter Reinholdtsen

weetmuts, 2022-01-12:
> Is there a way to find the key.

No, sorry the encryption is high security AES 128. You have to ask the meter owner, typicall your water company or municipality for the key.

petterreinholdtsen, 2022-02-09:
[Fredrik Öhrström]
>> Is there a way to find the key.
>
> No, sorry the encryption is high security AES 128. You have to ask the
> meter owner, typicall your water company or municipality for the key.

OK.  It worked. :)

--
Happy hacking
Petter Reinholdtsen

r0bertstewart, 2023-01-18:
So you’re in Australia? We are the Kamstrum distributor here. Give me a PM and I’ll see if we can help.

closed 2024-05-09 by xaelsouth