xaelsouth/rtl-wmbus #30: issue with missing telegrams form techem data4

MaxPeal, 2022-01-05:
i have a issue with missing telegrams form techem data4 

captured with 
```rtl_sdr -d 10smartV5 -f 868.625M -s 1600000 - | tee techem_data4_f868.625M_s1.6M.bin | rtl_wmbus -s```

at the same time i captured with a second sdr:
```rtl_433 -R 104 -R 105 -d :20Nano2Plus  -f 868900000 -s 1600000 -M protocol -M level -g "" -F kv -F json:rtl_433-test-out.json```
```tail -f rtl_433-test-out.json | jq '. | select( .id | contains(3719685))'```

```
wc -l rtl_433-test-out.json techem_data4_f868.625M_s1.6M.bin.decode
   276 rtl_433-test-out.json
   589 techem_data4_f868.625M_s1.6M.bin.decode
```
```
cat techem_data4_f868.625M_s1.6M.bin.decode | grep '3719685' | wc -l
1
cat rtl_433-test-out.json | jq -c '. | select( .id | contains(3719685))' | wc -l
3
```

```
cat techem_data4_f868.625M_s1.6M.bin.decode | grep '3719685' 
t2a;C1;1;1;2022-01-05 12:06:55.000;145;132;03719685;0x574468508596710394088c0085900f002c25a84a0900595833b45f776a5d7aa80030071026908a167e0db9fc2d258bab6f47e11f3131e60c151f24ad704661d8507d16e028eed2a245e18f244541bf4669a4d4b20fd96e95
```

```
cat rtl_433-test-out.json | jq -c '. | select( .id | contains(3719685))' 
{"time":"2022-01-05 13:01:10","protocol":104,"model":"Wireless-MBus","mode":"C","M":"TCH","id":3719685,"version":148,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":88,"data":"554468508596710394088c0085900f002c25a64a0900252af850cc21981b7aa6003007104017b3f10bcb4be16c8e9f97a9e2a8e00a9f7eab5619638e69bdf5bc76d1108b469460c528fa997edfe6af3201bc42ba0fd96e958025","mic":"CRC","CI":140,"AC":0,"ST":0,"CW":0,"max_volume_flow_min_1":"618122835248107.625 m3/min","inst_energy_wh_0":"10625320.000 kWh","inst_energy_j_1":"25.000 kJ","mod":"FSK","freq1":868.916,"freq2":868.993,"rssi":-0.102,"snr":17.171,"noise":-17.273}
{"time":"2022-01-05 13:01:43","protocol":104,"model":"Wireless-MBus","mode":"C","M":"TCH","id":3719685,"version":148,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":88,"data":"554468508596710394088c0085900f002c25a74a0900589baeb2d5365fdd7aa700300710fbf9b4432a1efd0712fe0f7fe1eea8b8b19bbe7aa6a216b50b6b7b915d2112b94a3a3bf12c61aa62bcf6f2f9871abc460fd96e95328c","mic":"CRC","CI":140,"AC":0,"ST":0,"CW":0,"max_volume_flow_min_1":"618122835248107.625 m3/min","inst_energy_wh_0":"10725320.000 kWh","inst_energy_j_1":"58.000 kJ","max_temperature_return_28":"1078143.000 C","inst_power_jh_0":"0.000 J/h","inst_volume_0":"575649494246685.250 mm3","unknown":"none","min_opertime_h_17":"639181711215.000 hours","min_volume_0":"-7.100 mm3","inst_volume_flow_h_1":"1514.100 mm3/h","min_temperature_diff_0":"1572263.100 K","err_volume_flow_h_19":"-99.690 m3/h","mod":"FSK","freq1":868.896,"freq2":868.985,"rssi":-0.111,"snr":17.133,"noise":-17.245}
{"time":"2022-01-05 13:02:16","protocol":104,"model":"Wireless-MBus","mode":"C","M":"TCH","id":3719685,"version":148,"type":8,"type_string":"Heat Cost Allocator","C":68,"data_length":88,"data":"554468508596710394088c0085900f002c25a84a0900595833b45f776a5d7aa80030071026908a167e0db9fc2d258bab6f47e11f3131e60c151f24ad704661d8507d16e028eed2a245e18f244541bf4669a4d4b20fd96e95604c","mic":"CRC","CI":140,"AC":0,"ST":0,"CW":0,"max_volume_flow_min_1":"618122835248107.625 m3/min","inst_energy_wh_0":"10825320.000 kWh","inst_energy_j_1":"59.000 kJ","max_power_jh_1":"0.000 kJ/h","mod":"FSK","freq1":868.888,"freq2":868.984,"rssi":-0.096,"snr":17.734,"noise":-17.831}```

MaxPeal, 2022-01-05:
```
$ rtl_wmbus -V
rtl_wmbus: d2be82c with local changes
d2be82cfa2180efb19937ad806dce62b5fe846a0 with local changes
```
```
$ git diff
diff --git a/build/rtl_wmbus b/build/rtl_wmbus
index 30484ee..86a90cc 100755
Binary files a/build/rtl_wmbus and b/build/rtl_wmbus differ
diff --git a/build/version.h b/build/version.h
index 732337e..5a41537 100644
--- a/build/version.h
+++ b/build/version.h
@@ -1,2 +1,2 @@
-#define VERSION "a07541f with local changes"
-#define COMMIT "a07541f904025f4c07062f81ef784205c3ef4ca8 with local changes"
+#define VERSION "d2be82c with local changes"
+#define COMMIT "d2be82cfa2180efb19937ad806dce62b5fe846a0 with local changes"
$ 
```
```
rtl_433 -V
rtl_433 version 21.12-18-gdee91e58 branch master at 202201031017 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
```

MaxPeal, 2022-01-05:
[techem_data4_f868.625M_s1.6M.bin.decode.log](https://github.com/xaelsouth/rtl-wmbus/files/7814903/techem_data4_f868.625M_s1.6M.bin.decode.log)
[rtl_433-test-out.json.log](https://github.com/xaelsouth/rtl-wmbus/files/7814904/rtl_433-test-out.json.log)

MaxPeal, 2022-01-05:
[techem_data4_f868.625M_s1.6M.bin.decode.log](https://github.com/xaelsouth/rtl-wmbus/files/7814903/techem_data4_f868.625M_s1.6M.bin.decode.log)
[rtl_433-test-out.json.log](https://github.com/xaelsouth/rtl-wmbus/files/7814904/rtl_433-test-out.json.log)
```
$ sha1sum techem_data4_f868.625M_s1.6M.bin
50c4cbd9b77b355b8eaa17fea270cfc9b8a87a61  techem_data4_f868.625M_s1.6M.bin

$ zcat techem_data4_f868.625M_s1.6M.bin.gz | split - -b 25m --filter='gzip > $FILE.gz' techem_data4_f868.625M_s1.6M.bin.part.
```
[techem_data4_f868.625M_s1.6M.bin.part.ad.gz](https://github.com/xaelsouth/rtl-wmbus/files/7815164/techem_data4_f868.625M_s1.6M.bin.part.ad.gz)
[techem_data4_f868.625M_s1.6M.bin.part.ac.gz](https://github.com/xaelsouth/rtl-wmbus/files/7815168/techem_data4_f868.625M_s1.6M.bin.part.ac.gz)
[techem_data4_f868.625M_s1.6M.bin.part.ab.gz](https://github.com/xaelsouth/rtl-wmbus/files/7815169/techem_data4_f868.625M_s1.6M.bin.part.ab.gz)
[techem_data4_f868.625M_s1.6M.bin.part.aa.gz](https://github.com/xaelsouth/rtl-wmbus/files/7815171/techem_data4_f868.625M_s1.6M.bin.part.aa.gz)

xaelsouth, 2022-01-08:
Hello Max,

thanks for very comprehensive description! I have joined your techem_data4.cu8 and checked with rtl_wmbus - got the same output as attached. That's good so far.

Some observations on your issue, please correct me if I'm wrong with something:

- you have used different RTL-USB-Dongles with rtl_wmbus and with rtl_433. I derived it from "-d 10smartV5" and "-d :20Nano2Plus"
- you have used different frequencies to receive at: 868.625M with rtl_wmbus and at 868.9M with rtl_433
- wireless-MBus device with the serial number 3719685 sends every 33 seconds (tis observation is based on searching for this device through rtl_433-test-out.json.log)
- you have used different observation periods: 65.43 seconds <= (sizeof(techem_data4_868.625M_1600k.cu8)/sample_rate=1600000) with rtl_wmbus and 100 seconds <= (13:02:21 - 13:00:41) with rtl_433

I think both result are not comparable, at least because of different dongles. A dongle (and antenna) has a strong impact on the receiving quaility!

Also I'm not sure about frequencies you receiving at: why do you use 868.9M for rtl_433? The right frequency were 868.95M. Why do you use 868.625M for rtl_wmbus, do you have to receive S1 mode datagrams simultaneously with T1/C1? If not then use 868.95M (option for rtl_sdr) and don't use the "-s" option for rtl_wmbus. It will switch  the frequency translation off (and reducing the CPU usage a little) and will prevent some rounding errors introduced by the frequency translation.

You should use the same observation periods. I would expect one or max. two datagrams within 65 seconds.

Xael

xaelsouth, 2022-01-16:
Closed this, no reaction from the questioner.

closed 2022-01-16 by xaelsouth