xaelsouth/rtl-wmbus #31: Package in Debian?

petterreinholdtsen, 2022-02-01:
Are you interested in having rtl-wmbus packaged in Debian? I've registered requests for packages for it and its reverse dependency in https://bugs.debian.org/998720 and https://bugs.debian.org/998721 and could help make it happen, if you are interested.  See also https://github.com/weetmuts/wmbusmeters/issues/460

weetmuts, 2022-02-01:
Yes! I am preparing for a deb package generation! Expect something in this or next week!

weetmuts, 2022-02-01:
Ah, I meant to post this comment on the wmbusmeters issue. @petterreinholdtsen If you can help out packaging rtl_wmbus that would be great! I will work on the wmbusmeters side.

petterreinholdtsen, 2022-02-01:
[Fredrik Öhrström]
> Ah, I meant to post this comment on the wmbusmeters
> issue. @petterreinholdtsen If you can help out packaging rtl_wmbus
> that would be great! I will work on the wmbusmeters side.

While I am happy to assist with the initial packaging, I lack the spare
time and bandwidth to take responsiblity for more packages, and thus
limit myself to sponsoring (aka supervise and help with the uploading).
Someone else need to take the leed on rtl_wmbus.

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2022-03-02:
Any luck?

petterreinholdtsen, 2022-04-28:
I tested making a Debian package by asking dh_make from dh-make to create a single package build framework, and discovered a patch like https://github.com/xaelsouth/rtl-wmbus/pull/33 is needed to get the build going and the binary to
install in the proper location.

xaelsouth, 2022-05-07:
Hi Petter,

just merged your patches into - thanks.

In debian/control I had to replace  Build-Depends: debhelper-compat (= 13) with ... (= 11) because I have  an old Ubuntu and having (= 13) did not work.  I'm not familiar with building of deb packages and hope that I did not broke something ;)

Xael

petterreinholdtsen, 2022-05-07:
[Xael South]
> In debian/control I had to replace  Build-Depends: debhelper-compat (=
> 13) with ... (= 11) because I have  an old Ubuntu and having (= 13)
> did not work.  I'm not familiar with building of deb packages and hope
> that I did not broke something ;)

Most likely did not break anything.  The version control debhelper
behaviour, and you can check under 'COMPATIBILITY LEVELS' in
debhelper(7) what changed from 11 to 13.  I had a quick look to see if
anything seemed relevant, without noticing anything.

-- 
Happy hacking
Petter Reinholdtsen