xaelsouth/rtl-wmbus #38: Test with IZAR RC 434 I R4

almerindo, 2022-08-06:
I would like to try with the module [IZAR RC 434 I R4](https://www.diehl.com/metering/en/products-services/system-software/system-software-components/izar-rc-i-r4-at/)  **model - 434MHz.**

The following command didn´t work:
rtl_sdr -f 434M -s 1600000 - 2>/dev/null | build/rtl_wmbus

Can you help me?

xaelsouth, 2022-08-18:
Hello almerindo,

I'm sorry, but what you want to do - it will not work. According to
https://www.diehl.com/metering/en/products-services/system-software/system-software-components/izar-rc-i-r4-at/ your device is sending in R-mode (R3 or R4?) WmBus mode, which is not implemented now.

FYI: Appendix of https://www.ti.com.cn/cn/lit/an/swra423/swra423.pdf contains a nice summary of  all possible OMS (WmBus) modes.

Xael

almerindo, 2022-08-23:
Hi!  I tried with the right frequency 434.47 and It is working now!

rtl_sdr -f 434.47M -s 1600000 - 2>/dev/null | ./build/rtl_wmbus

xaelsouth, 2022-08-24:
Ок, I thought that this water meter would be sending consumption values in R Mode, which is currently not supported by rtl-wmbus. Bit it was a "usual" T1 Mode. Sorry for having put some irritations into it ;)

closed 2024-05-09 by xaelsouth