xaelsouth/rtl-wmbus #39: Honeywell V210 Water Meter 868MHz T1

jamescarter, 2022-08-18:
Would we expect [this device](https://www.mwatechnology.com/wp-content/uploads/2018/02/ELS10098_V200_210_Specsheet_V1_BM.pdf) to work?

The specsheet mentions a configurable 12 second transmit interval and in sdr++ I do see transmissions every 3-10 seconds (which I presume are from the same meters installed by my neighbours).

This command does not return any results:
`rtl_sdr -f 868.625M -s 1600000 - 2>/dev/null | build/rtl_wmbus -s`

In sdr++ transmissions appear to start at exactly 868M (which I suppose could be something else). Using `-f 868M` makes not difference.

Would love a pointer in the right direction and happy to help try and contribute adding support if it's not already

xaelsouth, 2022-08-18:
Hello James.

Well, T1 mode means your device is sending at 868.95M. I suggest you to try with

rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | build/rtl_wmbus

You probably have to play in 20..25 kHz steps with "-f" option around the center frequency (868.95M) . Some RTL-SDR-Dongles can't tune precisely enough to that freq and must be tuned manually to it. 

Xael

jamescarter, 2022-08-19:
Hi Xael, thanks a lot for the super quick response! I've been playing around with a few things based on you help

When I check out a public FM frequency `99.7M` it's perfectly centered, but perhaps that's because it's a strong signal.

Since you confirmed I should be looking at `868.95M` I see literally nothing there, but instead see between:
- 868.44 - 868.56 (lines)
- 869.4 - 869.65 (squares)

[image]

I've been playing around with the frequencies (manually) andalso  wrote a little script to help test however I'm not getting any success. I'm starting to wonder if these are wmbus signals or not now.

I'm thinking perhaps I need to isolate the problem - create a faraday cage around my antenna and sit on-top of my water meter and see if that picks anything up.

FYI this is the simple script, it listens for 14s before moving onto the next frequency off center (I've tried in 25 and 5kHz steps too)
```
#!/bin/bash
BASE_FREQ_HZ=868950000
SAMPLE_RATE_HZ=1600000
GAIN_DB=20
STEP_HZ=20000
TIMEOUT=14s

echo "Base:      with frequency      $BASE_FREQ_HZ"
timeout $TIMEOUT rtl_sdr -f $BASE_FREQ_HZ -s $SAMPLE_RATE_HZ -g $GAIN_DB - 2>/dev/null | rtl_wmbus

for i in {1..15}
do
  freq_up=$(($BASE_FREQ_HZ + ($STEP_HZ * $i)))
  freq_dn=$(($BASE_FREQ_HZ - ($STEP_HZ * $i)))

  echo "Step: $i with frequency up   $freq_up"
  timeout $TIMEOUT rtl_sdr -f $freq_up -s $SAMPLE_RATE_HZ -g $GAIN_DB - 2>/dev/null | rtl_wmbus
  
  echo "Step: $i with frequency down $freq_dn"
  timeout $TIMEOUT rtl_sdr -f $freq_dn -s $SAMPLE_RATE_HZ -g $GAIN_DB - 2>/dev/null | rtl_wmbus
done
```

pswiatki, 2022-08-20:
When you say 99.7 FM is

> perfectly centered

what do you mean? 
Check the FFT bin size and video bandwidth resolution. Perhaps when you zoom in on the peak you will see some (slight?) offset you could apply in your wmbus tests? Also, I believe wmbus transmissions will not be exactly at the centre frequency and always be offset by a known value (not sure if this may be thought of as a particular channel the device is TXing on)  - which is the reason the channel decoder must take a wider chunk of spectrum around the centre freq.

pswiatki, 2022-08-20:
When you say 99.7 FM is

> perfectly centered

what do you mean? 
Check the FFT bin size and video bandwidth resolution. Perhaps when you zoom in on the peak you will see some (slight?) offset you could apply in your wmbus tests? Also, I believe wmbus transmissions will not be exactly at the centre frequency and always be offset by a known value (not sure if this may be thought of as a particular channel the device is TXing on)  - which is the reason the channel decoder must take a wider chunk of spectrum around the centre freq.

If the above is true, a slight offset of your RX device should not matter much - I am pretty sure the decoder detects and corrects for this offset providing it doesn't exceed some limit (say, channel boundary).

jamescarter, 2022-08-20:
@pswiatki no doubt I'm showing my lack of experience using that 

[image]

pswiatki, 2022-08-20:
Yeah, from the picture above it sure looks like nicely centred. Thing is: 1 tick on the freq axis is 100 kHz. If your offset was, say, 5 kHz - could you actually see it (given that it is a live transmission and spectrum is dynamic, changes all the time.
But then again: would such offset matter for the `wmbus` decoder? I posit: no, it would not.

closed 2022-08-22 by jamescarter

xaelsouth, 2022-08-22:
Hello James.

1) Check if the radio is enabled by your water supplier at all.  It' s not always the case! Your device has an another (pulse) interface for reading consumption values, but it's unlikely going to be used.

2) your supplier can still read consumption  values just by walk in and writing down the reading value on the paper :) Or you have to tell him values monthly or yearly by phone or via customer portal.

3) do you have any reading infrastructure there,  like data collectors? I have seen use cases, in which water meters have sent data to installed data collectors only twice in a month - on 1th and 16th :)