xaelsouth/rtl-wmbus #41: RSSI value in dBm

pbasista, 2023-03-28:
Hi,

I have noticed that the RSSI values reported by rtl-wmbus are always positive. Packets received via weak signals have RSSI of around 20 or less. Packets received with strong signals have RSSI as high as 130 or more.

Is there a reasonable way to convert these values to decibel-milliwatts ([dBm](https://en.wikipedia.org/wiki/DBm))?

pswiatki, 2023-03-28:
I guess there is no way that is reasonable. You see: AGC is enabled as far as I know and it changes as necessary during the reception.

xaelsouth, 2023-05-08:
Hello pbasista,

you can calculate dBm instead of RSSI by using following formula even it doesn't make much sense, see below:

dBm = 10 * log10[ (I^2 + Q^2) / (2 * R * 0.001) ]

Where R is receiver's input impedance, which is typically 75 ohms for DVB-T receivers (some of them have got 50 ohms!), referenced to 0.001W.

This formula can be applied **only if** no AGC comes into play, as pswiatki has already mentioned. Digital receivers usually take AGC into account and put a **corrected** RSSI and/or dBm value into their registers.

Xael

closed 2023-05-08 by xaelsouth