xaelsouth/rtl-wmbus #44: Wrong behaviour on stadout

alvaromemo, 2023-08-17:
Running this command was working great

`rtl_sdr -f 434.475M -s 800000 - 2>/dev/null | /usr/bin/rtl_wmbus -p s`

But suddenly starts showing this.

[image]

Running this command

`wmbusmeters rtlwmbus:434.475M:t1 Sappel izar ANYID ANYKEY`

Shows this.

[image]

Why this happens and how i restart the rtl_wmbus service?

alvaromemo, 2023-08-17:
I have to reinstall `wmbusmenters` and works again. Any ideas?.

xaelsouth, 2024-05-09:
Hello alvaromemo.

No ideas, but you are trying to use rtl_sdr and rtl_wmbus in a strange way:
rtl_sdr -f 434.475M -s 800000 - 2>/dev/null | /usr/bin/rtl_wmbus -p s

You have given the sample rate as 800000. In that case you need to pass "-d 1" to rtl-wmbus.
And you have passed a "s" argument, you probably wanted "-s", but do you really want to receive S1, T1 and C1 at 434MHz?..

closed 2024-05-09 by xaelsouth