xaelsouth/rtl-wmbus #45: rtl_wmbus cannot read from rtl_sdr -- how to debug -- or can it use rtl_433 instead ?

stbuerger, 2023-11-21:
Hello - I' m facing a strange problem with rtl_wmbus. The facts are ..
- rtl_433 correctly receives and decodes transmission from my Kamstrup Multical 21 water meter
  `rtl_433 -f 868.625M -s 1600000 -R 104 -R 105 `
- `rtl_sdr -f 868.625M -s 1600000` produces _some_ output 
- `rtl_sdr -f 868.625M -s 1600000 | rtl_wmbus -s` doesn't produce any output
- `cat samples.bin | rtl_wmbus` works fine (samples.bin from repo/source)

The question I now have
1) How can I further debug this ? How can I "validate" output from rtl_sdr ?
2) Could I somehow use "raw" data from rtl_433 to feed to rtl_wmbus ?

I know I can use rtl_433 as device/input for wmbusmeters (and got it working) but would prefer to use rtl_wmbus.
(Can't get correct RSSi from rtl_433)

thx a lot

-sb

xaelsouth, 2023-12-04:
... and make sure you have put '-'  (a minus) which means that rtl_sdr will redirect its output to stdout!

xaelsouth, 2024-05-09:
I nothing worked until now, please check out with the new version of rtl_wmbus. I'm gonna close this issue.

closed 2024-05-09 by xaelsouth