xaelsouth/rtl-wmbus #54: No telegrams for apator e-itn 30.4

spbdimka, 2025-01-22:
good day, I'm trying to get data from heat sensors. I have at least 6 of them, it's an apartment building
I'm using a blue Chinese sdr - dvb-t+fm+dab (R860 & SDR)

I tried it on both Windows and Linux

> rtl_sdr -f868.625M -s1800000 -g20 - | rtl_wmbus -o -s
> Found 1 device(s):
> 0: Realtek, RTL2838UHIDIR, SN: 00000001
> 
> Using device 0: Generic RTL2832U OEM
> Found Fitipower FC0012 tuner
> Sampling at 1800000 S/s.
> Tuned to 868625000 Hz.
> Tuner gain set to 19.20 dB.
> Reading samples in async mode...

I tried adjacent frequencies with a step of 50 kHz, sampling frequency of 1600000 and launch with different keys
zero telegram, at the same time in sharpsdr I periodically see information on these frequencies
I can't understand what the problem is - my receiver or the sensors are silent (according to the specification, they should report information once every 4 minutes and once every 30 seconds for three days on the day of transmitting readings)

Please help with directions to solve the problem

xaelsouth, 2025-01-25:
Good day. The sampling frequency has been set wrong. Choose 1**6**00000 instead of 1**8**00000. If it does not work, remove "-g20" from the rtl-sdr cmdline.

spbdimka, 2025-01-26:
no luck
g20 if I'm not mistaken, should help improve the signal, setting amplifier to max, anyway not telegrams

xaelsouth, 2025-02-12:
1. But you have tried with the correct sampling rate of 1600000? I'f asking because you had put 1800000 in the command line.
2. Try to remove -o from rtl-wmbus's command line. Typically there is no need of using DC filter
3. Or try at least jut this: rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | build/rtl_wmbus

As the Apator HCA operates on 868.95MHz it should work. Is the radio enabled?...

spbdimka, 2025-02-12:
Sure, im also buy dongle on r820t instead r860
Same result